AdrianEddy/telemetry-parser #1: Missing GPS info?

rgaufman, 2021-09-11:
This is great, but the output CSV seems to be missing a lot of information like GPS?

I can see GPS there in the output, but it seems to be thrown away in the csv output?

```
$ ./gyro2bb --dump GOPR1417.MP4
...
, z: 139 }, Vector3 { x: 4191, y: -2049, z: 311 }, Vector3 { x: 4307, y: -1731, z: 242 }, Vector3 { x: 4649, y: -1681, z: 111 }, Vector3 { x: 4693, y: -1613, z: 119 }, Vector3 { x: 4521, y: -1271, z: 121 }, Vector3 { x: 4240, y: -517, z: 166 }, Vector3 { x: 4005, y: 494, z: 125 }, Vector3 { x: 3944, y: 1021, z: 375 }, Vector3 { x: 4137, y: 876, z: 597 }, Vector3 { x: 4381, y: 788, z: 612 }, Vector3 { x: 4280, y: 480, z: 646 }, Vector3 { x: 4069, y: 99, z: 456 }, Vector3 { x: 3973, y: -255, z: 206 }, Vector3 { x: 3820, y: -711, z: 183 }, Vector3 { x: 3711, y: -991, z: 125 }, Vector3 { x: 3350, y: -1224, z: 430 }, Vector3 { x: 2934, y: -1237, z: 655 }, Vector3 { x: 2743, y: -1160, z: 912 }, Vector3 { x: 2715, y: -946, z: 1156 }, Vector3 { x: 3026, y: -953, z: 844 }, Vector3 { x: 3512, y: -1071, z: 658 }, Vector3 { x: 3864, y: -1148, z: 770 }, Vector3 { x: 4381, y: -1092, z: 390 }, Vector3 { x: 5108, y: -1093, z: -147 }, Vector3 { x: 5309, y: -1126, z: -237 }, Vector3 { x: 5111, y: -1098, z: -134 }, Vector3 { x: 5134, y: -978, z: -488 }, Vector3 { x: 5243, y: -724, z: -920 }, Vector3 { x: 5143, y: -436, z: -1199 }, Vector3 { x: 4680, y: -220, z: -1299 }, Vector3 { x: 3989, y: -102, z: -1101 }, Vector3 { x: 3760, y: 36, z: -966 }, Vector3 { x: 3881, y: 290, z: -981 }, Vector3 { x: 4169, y: 472, z: -1032 }, Vector3 { x: 4227, y: 324, z: -963 }, Vector3 { x: 4136, y: 46, z: -823 }, Vector3 { x: 4028, y: -211, z: -542 }]
Accelerometer             Unit                      SIUN                                              : m/s²
Accelerometer             Temperature               TMPC                                              : 37.25
Accelerometer             Scale                     SCAL                                              : 418
Accelerometer             Count                     TSMP                                              : 6500
Accelerometer             0x5449434b                TICK                                              : 98377
GPS5                      Name                      STNM                                              : GPS (Lat., Long., Alt., 2D speed, 3D speed)
GPS5                      Unit                      UNIT                                              : degdegmm/sm/s
GPS5                      Scale                     SCAL                                              : [10000000, 10000000, 1000, 1000, 100]
GPS5                      Count                     TSMP                                              : 585
GPS5                      0x47505335                GPS5                                              : [[98366205, 1233808163, 78862, 14149, 1410], [98366275, 1233808173, 78873, 14185, 1415], [98366345, 1233808182, 78890, 14212, 1419], [98366414, 1233808191, 78903, 14208, 1422], [98366483, 1233808201, 78919, 14128, 1421], [98366553, 1233808209, 78935, 14158, 1413], [98366624, 1233808220, 78977, 14254, 1416], [98366695, 1233808230, 79003, 14318, 1426], [98366766, 1233808239, 79036, 14305, 1432], [98366835, 1233808248, 79038, 14227, 1431], [98366904, 1233808259, 79054, 14208, 1423], [98366974, 1233808268, 79069, 14187, 1421], [98367044, 1233808277, 79083, 14206, 1419], [98367114, 1233808287, 79100, 14273, 1421], [98367185, 1233808297, 79108, 14368, 1428], [98367256, 1233808308, 79113, 14398, 1437], [98367326, 1233808317, 79120, 14347, 1440], [98367396, 1233808326, 79143, 14277, 1435], [98367466, 1233808336, 79166, 14273, 1428]]
GPS5                      0x47505346                GPSF                                              : 3
GPS5                      0x47505350                GPSP                                              : 453
GPS5                      0x47505355                GPSU                                              : 2017-12-11 09:06:18.345 UTC
GPS5                      0x5449434b                TICK                                              : 98400
ISOG                      Name                      STNM                                              : Sensor gain (ISO x100)
ISOG                      Count                     TSMP                                              : 806
ISOG                      0x49534f47                ISOG                                              : [2.9510086, 2.938307, 2.938307, 2.938307, 2.9215407, 2.9215407, 2.9215407, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644, 2.9049644]
ISOG                      0x5449434b                TICK                                              : 98401
Done in 159.480 ms
```

AdrianEddy, 2021-09-12:
Currently only gyroscope and accelerometer is used for the csv. I'll add GPS output soon.

jeroentbt, 2024-07-23:
I'll add my two cents here :)

My usecase is mainly about action sports.
First, I want to saveguard the telemetry data of my gopro footage before messing with the video files themselves. That seems covered by using `--dump`.

I upload the GPS data from my GoPro video to [serious racing](https://serious-racing.com/), they handle extracting the GPS data through mp4box.js. If you telemetry parser would include gps data in the csv, I could upload that.

I could use this same file to create data overlays.
Most of the apps (eg [racerender](https://racerender.com/RR3/Features.html), [telemetry overlay](https://goprotelemetryextractor.com/karting-motorbike-rally-trackday-overlay), ...) to create these overlays do support extracting the data from the video, though. If I have the CSV, I could use that.

I am starting to collect other data points (through an ODB2 connector) and will likely need to work with and combine the data myself to present something useful for analysis or more intricate visualisation.

So no real need actually, for me.
Having the GPS data in the CSV could potentially make it a bit easier to work with the data myself.

It comes down to a thank you :)
Thank you for taking the time to create this nice piece of software!

DuVogel87, 2024-12-23:
How is the progress on that so far?

tomilepp, 2025-01-07:
Hi, i made small proof of concept program to extract GoPro GPS data: https://github.com/tomilepp/gopro-gps

It reads GPS data and output it to kml format.