AdrianEddy/telemetry-parser #18: request: DJI Action 2

gretel, 2022-07-30:
have noticed https://github.com/AdrianEddy/telemetry-parser/blob/master/src/dji/mod.rs - can you please elaborate on the current state of this? i'm asking from the `gryoflow` perspective 😼 

```
./gyro2bb /Users/tom/Desktop/DJI_0005.MP4
Detected camera: DJI Action 2
Done in 4658.201 ms

./gyro2bb /Users/tom/Desktop/DJI_0008.MP4
Garbage found at the end of the file, removing 2098408 bytes from the end.
Detected camera: DJI Action 2
Done in 14480.611 ms
```

any value in sharing the result? regards

AdrianEddy, 2022-07-31:
Currently Action 2 doesn't record gyro data, we need to wait for DJI to update firmware to include this data.

The current parser was an initial guesswork to figure out the debug stream and see if there is gyro data, but apparently there isn't

closed 2022-07-31 by AdrianEddy

gretel, 2022-08-03:
> Currently Action 2 doesn't record gyro data, we need to wait for DJI to update firmware to include this data.
> The current parser was an initial guesswork to figure out the debug stream and see if there is gyro data, but apparently there isn't

i get it.. too bad for now. thanks for the feedback!