AdrianEddy/telemetry-parser #22: Windows version of the app does not open

Slint7, 2022-12-22:
Good day

I have Windows 10. I tried the 32 & 64 versions.

When I double-click the .exe file, nothing happens.

Do you have an explanation on how to use the telemetry-parser?

Am I using it incorrectly, do I need a prerequisite or can Windows security be blocking the app?

Thanks for the great work.

closed 2022-12-22 by AdrianEddy

Slint7, 2022-12-22:
> it's a command line application, it doesn't have a graphical user
> interface.
> What do you want to use it for? If you just drag & drop a video file over
> gyro2bb.exe it will produce .csv by default

Thank you for the explanation.

Slint7, 2022-12-22:
When I drag Gopro footage to the .exe it works fine however, I can't get
the Open Camera Sensors .csv files working.

How does it work with the Andoid app, Open Camera Sensors?

(The Open Camera Sensor files are 4 different .csv files for the
gyro, accelerometer, timestamps and magnetic info)

I want to use the Open Camera Sensor data with Gyroflow video stabilization.

AdrianEddy, 2022-12-22:
You don't have to use this app at all then, just open the files directly in Gyroflow v1.4.0