AdrianEddy/telemetry-parser #31: Question: Scale Calculation Accelerometer

Ryan37342342, 2024-02-16:
"Accelerometer": {
            "Name": "Accelerometer",
            "Data": "[Vector3 { x: 2523, y: 1871, z: 4855 }, Vector3 { x: 1787, y: 1191, z: 4347 }, Vector3 { x: 807, y: -11, z: 2759 }, Vector3 { x: 
            "Unit": "m/s\u00b2",
            "Matrix": "[[0.0, 0.0, 1.0, 0.0, -1.0, 0.0, 1.0, 0.0, 0.0]]",
            "Scale": "418",
            "Count": "231",
            "Orientation": "ZyX",
            "OrientationIn": "YxZ",
            "OrientationOut": "ZXY"
        },
Hi bit of a dumb question but I need clarification on the following point, 
How does scale relate to the values in data ? ie how to get z in ms/^2 ?
I am using a gopro
Cheers

AdrianEddy, 2024-02-16:
I'll need more details about what you're trying to do, but in general you take the value (eg 2523) an divide by scale (418) to get the unit (m/s2)

Ryan37342342, 2024-02-16:
Sweet as, thats what I wanted to know cheers

closed 2024-02-16 by Ryan37342342