AdrianEddy/telemetry-parser #32: Issue with Sony RX0 MII

gabrielecoppi, 2024-06-06:
I am trying to get the gyro and accelerometer data out of a video taken with the Sony. However, I am getting this error (using the python wrapper installed with pip). 

```
thread '<unnamed>' panicked at /home/runner/work/telemetry-parser/telemetry-parser/src/sony/mod.rs:109:95:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
pyo3_runtime.PanicException: called `Option::unwrap()` on a `None` value
```

I have no real clue on what is happening here, The video lives [here](https://drive.google.com/file/d/1dFSdBo1-_qOl1yvyDwVjr7B5_n0OewdB/view?usp=sharing)

AdrianEddy, 2024-06-06:
Does gyro2bb cli tool read it correctly?

gabrielecoppi, 2024-06-06:
This is what I am getting using directly `gyro2bb`
```
>> ./gyro2bb ~/Desktop/C0006.MP4 
Detected camera: Sony DSC-RX0M2
thread 'main' panicked at /home/gabriele/Documents/telemetry-parser/src/sony/mod.rs:41:9:
assertion failed: v.len() == 3
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

gabrielecoppi, 2024-06-06:
I did a bit of more research and if I use the `-d` option on gyro2bb I get the data up to a point and then get 

```
LensOSS                   0xe410                    Lens OSS position/rotation 3xi32                  : 0 0 8124
thread 'main' panicked at /home/gabriele/Documents/telemetry-parser/src/sony/mod.rs:130:95:
called `Option::unwrap()` on a `None` value
```

AdrianEddy, 2024-06-06:
looks like parts of this file are damaged. I updated the code to ignore these errors instead of panicking, and it appears to work fine now. You can [compile the python module from source](https://github.com/AdrianEddy/telemetry-parser/tree/master/bin/python-module#building), let me know if that works for you

AdrianEddy, 2024-06-06:
If you're using the `normalized_imu()` function then yes, it's deg/s and m/s² for accl

closed 2024-06-06 by AdrianEddy