AdrianEddy/telemetry-parser #34: Spec of the generated CSV file (e.g. units) ?

vdeconinck, 2024-10-12:
Hi,
No bug, just lacking some info :-)
I'm looking for a way to overlay an inclinometer on my offroad videos (part GoPro8, part Insta360 Ace) and would rather make it open source than buying commercial software. Rendering is working but I'd like to improve the data gathering. I first used gopro-utils (for GoPro) and exiftool (for Insta360) but it seems your telemetry-parser is a "one size fits all" solution. Nice! :-)
However, I'm using accelerometer only until now and it's heavily influenced by the fact the vehicle is moving. So I'd like to fusion with gyro data, but I have a hard time feeding a Kalman filter to get better roll and pitch angles (I'm a newbie in IMU :-)).
Could you tell me the units used in the file? I can't find the spec for the betaflight blackbox CSV...
Any other hint on how to perform that operation?
Thanks and keep on the good work!
Vincent

AdrianEddy, 2024-10-13:
the gyro values are in deg/s and the accelerometer you have to divide by 2048 to get m/s2

But you'll probably want to take a look at gyroflow's cli option [--export_metadata](https://github.com/gyroflow/gyroflow/blob/master/src/cli.rs#L72) which will give you already integrated and synchronized quaternions

closed 2024-10-13 by AdrianEddy

vdeconinck, 2024-10-13:
Hi.
Wow, thanks for the fast and comprehensive reply !
I'll make sure to check the export through gyroflow.
Thanks a ton and have a nice week !
KR, Vincent

vdeconinck, 2024-10-20:
Sorry to bother you again :-(
I installed giroflow from the release page (impressive piece of software !), but the export_metadata option seems to be more recent and no included in the latest release, right ? So I tried to build it myself following the instructions. Had a few hiccups but I can start gyroflow UI with "just run". 
However, if I do a "just run source.mp4 --extract_metadata telemetry.txt", I'm getting:
```
...
16:38:53 [DEBUG] (1) gyroflow: Gyroflow 1.5.4 (dev144874)
Unrecognized argument: --export_metadata
Run gyroflow.exe --help for more information.
```
Is there something I'm missing ?
Or is the cli a separate executable ?
KR,
Vincent

AdrianEddy, 2024-10-20:
Simply download the dev build from https://gyroflow.xyz/devbuild/

vdeconinck, 2024-10-20:
Wow, that's a blazingly fast answer :-) !
OK, I blindly copied the option from your post above, should have read the help :-/
It works with my build now, but I'll grab the dev build from gyroflow.xyz for ease of use.
Thanks again very much for your time and help.

vdeconinck, 2024-12-22:
Hi,
Just wanted to say thanks again for your software and help.
I finalized my small project and just published my first video with an inclinometer overlay based on angles extracted by gyroflow: https://youtu.be/5syqOH3W1C0?t=47 
Have a nice day.
Vincent