AdrianEddy/telemetry-parser #36: Real-time Processing Question

Leon-Williams9, 2024-12-04:
Hi, great work on this project! I have a question—would it be possible to adapt this algorithm for real-time processing? I’m trying to synchronize it with other sensors and would appreciate any tips or suggestions you might have.

Thanks in advance!

AdrianEddy, 2024-12-04:
what do you mean? there's no algorithm here, it's just a parser.
If you have a real-time stream of metadata - what format is it in? 
if you're asking if parser for your custom streamed metadata can be added here then probably yes, but I'll need more details

Leon-Williams9, 2024-12-04:
The IMU data should be associated with the video data. For example, with an Insta360 camera, which is a rolling-shutter type camera, how can I synchronize its data with the IMU and LiDAR timestamps? This is similar to the scenario described in [this post](https://community.theta360.guide/t/sensor-fusion-not-necessarily-in-real-time-sync-2-ricoh-theta-v-with-a-lidar/7427/6).

[image]

AdrianEddy, 2024-12-04:
depends on what is sent together with the IMU data. there may be simply a time offset
Insta360 stored metadata (in mp4 files) has a field called `gyro_timestamp` which is the offset between video frame and gyro data

It's impossible to tell without seeing actual streamed data first though

Leon-Williams9, 2024-12-04:
Thanks a lot for the helpful info! I’ll check the data format and analyze the possibilities. I understand it’s hard to be sure without the actual data, but your input is appreciated.