AdrianEddy/telemetry-parser #39: Question Regarding Output Location

DuVogel87, 2024-12-23:
I have a question regarding the output location when using the --dump and --imuo xZy commands. In the --help documentation, I couldn't find any information about where the results are saved.

Could you please clarify where the output is stored when I run these commands?

Thank you for your help!

AdrianEddy, 2024-12-23:
it is only printed to the terminal. To save it to file, use standard command line redirection, like `gyro2bb --dump file.mp4 > metadata.txt`

alternatively you can export the metadata to json in Gyroflow using `Motion Data -> Export` (or use the gyroflow cli to do that in command line)

closed 2024-12-23 by AdrianEddy

DuVogel87, 2024-12-23:
Thanks. 

Do you happen to have any idea how I can integrate the metadata into Agisoft Metashape?

AdrianEddy, 2024-12-23:
I have no idea, never heard of this software